LFiveAV/RC-car #6: Find compatible sensors

LFiveAV, 2019-04-24:
Create short lists (~3 options) for the following sensor types:

- Camera
- Ultrasonic
- Wheel encoders

Take into account the expected performance, price, connector types, power consumption, size etc.

junjiel, 2019-04-25:
I will start to look for wheel encoders first.

DanielTorres2, 2019-04-25:
Looking for camera

DanielTorres2, 2019-04-26:
CAMERAS:

Zed camera - https://www.stereolabs.com/zed/
USB 3.0 port
5 V/380mA
175 x 30 x 33 mm
 90° (H) x 60° (V) x 110° (D) 
1080p/@30 fps/3840x1080
159 g
$449

Is used in several RC car projects inclduing AVL Turkey. Great field of view, resolution and frame rate. Is fairly expensive.

STEEReoCAM - https://www.e-consystems.com/nvidia-cameras/jetson-agx-xavier-cameras/stereo-camera.asp#key-features
Plugs in to NVIDIA® Jetson AGX Xavier™/TX2 developer kit using  2-lane MIPI CSI-2 interface.
3.3 V/0.75W
77.9⁰(D), 54⁰(H), 49.5⁰(V)
((2*1600) x 1300) resolution at 22 fps
143.5 x 27.5 x 29 mm
161g
$299

Apparently,  MIPI CSI-2 interface is more efficient in processing camera inputs compared to USB ports. Short horizontal field of view and may be considered expensive.

Logitech C920 - https://support.logitech.com/en_us/product/hd-pro-webcam-c920/specs
USB 2.0
Power consumption?
(D) 78°, (H) 70.42°, 43.3°(V)
1080p@30fps
94 (W) mm x 24 (D) mm x 29 (H) mm
162g
790 kr

Cheap, small, decent frame rate and field of view. USB 2.0 is said to be one of the slowest processing interfaces and may be limited to 720p @30fps according to https://elinux.org/Jetson/Cameras.

Intel® RealSense™ Developer Kit (R200) - https://software.intel.com/sites/default/files/managed/d7/a9/realsense-camera-r200-product-datasheet.pdf
USB 3.0
1080p at 30 FPS
0.75 - 0.92 W/ Volt?
130mm x 20mm x 7mm
43 (V) 70 (H) 77 (D)
79$

Similar to the Logitech C920 but uses USB 3.0.

There are most likely even cheaper options available that are yet to be confirmed compatible with the TX2 such as Logitech C210 (135 kr) that is confirmed to work with TK1. Personally, I would go with either the ZED stereo or some web camera bellow 800 kr.

junjiel, 2019-05-02:
**Ultrasonic Distance Sensor - HC-SR04:**

Description:
This is the HC-SR04 ultrasonic distance sensor. This economical sensor provides 2cm to 400cm of non-contact measurement functionality with a ranging accuracy that can reach up to 3mm. Each HC-SR04 module includes an ultrasonic transmitter, a receiver and a control circuit.

There are only four pins that you need to worry about on the HC-SR04: VCC (Power), Trig (Trigger), Echo (Receive), and GND (Ground). 

Features:

- Operating Voltage: 5V DC

- Operating Current: 15mA

-  Measure Angle: 15°

- Ranging Distance: 2cm - 4m

Price: 3.95$

Link:
https://www.sparkfun.com/products/13959

closed 2019-05-02 by junjiel