LFiveAV/RC-car #16: Estimate power consumption

LFiveAV, 2019-05-02:


junjiel, 2019-05-02:
TX2: 7.5W, [https://devblogs.nvidia.com/jetson-tx2-delivers-twice-intelligence-edge/](url)
Zed camera: 5V*380mA = 1.9W, [https://www.stereolabs.com/zed/](url)
rp lidar: 5V*400mA = 2W, [https://www.robotshop.com/media/files/pdf2/rpk-02-datasheet.pdf](url)
IMU: [file:///C:/Users/u22b00/Downloads/BST_BNO055_DS000_14.pdf](url)

krlhrnbrnt, 2019-05-02:
GPS: 5.5V25mA = 0.14W https://www.adafruit.com/product/746
Ultrasonic: 5V15mA = 0.075W https://www.mouser.com/ds/2/813/HCSR04-1022824.pdf 
Arduino nano: 7-12V19mA < 0.228W  https://store.arduino.cc/arduino-nano 
Encoder: 5V20mA = 0.1W http://www.tinyosshop.com/index.php?route=product/product&product_id=541

closed 2019-05-06 by LFiveAV